etf-unibl/pds-2024 #22: Create 2-4 decoder with enable input (first method)

knezicm, 2024-10-28:
Potrebno je realizovati dekoder 2-4 sa ulazom za omogućenje. Više informacija o samom kolu, može se pronaći na sljedećem linku:

https://www.tutorialspoint.com/digital_circuits/digital_circuits_decoders.htm

Entitet dekodera treba da ima sljedeći izgled:

```vhdl
entity decoder_2_4 is
    port (i_A : in std_logic_vector(1 downto 0);
          i_E : in std_logic;
          o_Y : in std_logic_vector(3 downto 0));
end decoder_2_4;
```
gdje je `i_A` ulazni 2-bitni broj, `i_E` ulaz za omogućenje dekodera, a `o_Y` 4-bitni izlazni podatak.

Kolo realizovati opisom u VHDL jeziku korišćenjem **uslovnih konkurentnih naredbi**.

rankomirosav, 2024-10-31:
Done.

knezicm, 2024-11-12:
5 bodova

closed 2024-11-12 by knezicm