etf-unibl/pds-2024 #39: Create dual-code priority encoder

knezicm, 2024-11-12:
Realizovati prioritetni koder koji na izlazu postavlja dvije kodne riječi: jednu najvišeg prioriteta i drugu sljedećeg najvišeg prioriteta. Ulaz kola je 8-bitni signal `req`, a izlazi `code1`, `code2`, `valid1` i `valid2`, koji predstavljaju 3-bitne kodne riječi, odnosno 1-bitne signale za oznaku validnosti kodnih riječi najvišeg i drugog po redu najvišeg prioriteta. Entitet kola treba da ima sljedeći izgled:

```vhdl
entity dual_code_priority_encoder is
  port (
  REQ_i    : in  std_logic_vector(7 downto 0);
  CODE1_o  : out std_logic_vector(2 downto 0);
  CODE2_o  : out std_logic_vector(2 downto 0);
  VALID1_o : out std_logic;
  VALID2_o : out std_logic
);
end dual_code_priority_encoder;
```
gdje je `REQ_i` ulazni 8-bitni signal, `CODE1_o` i `CODE2_o` izlazne kodne riječi najvišeg prioriteta, a `VALID1_o` i `VALID2_o` signali koji označavaju validnost ovih kodnih riječi.

Za realizovano kolo napisati odgovarajući samoprovjerljivi (*self-checking*) *testbench* neophodan za simulaciju kola. Naziv VHDL fajla za *testbench* treba da bude u formatu *nazivDizajna_tb.vhd*.

Chicha13, 2024-11-20:
Potvrda o provjerama nakon pokretanja testova u lokalnom okruženju.
[log.txt](https://github.com/user-attachments/files/17825607/log.txt)

knezicm, 2025-01-29:
5 bodova

closed 2025-01-29 by knezicm