etf-unibl/pds-2024 #40: Create circuit for converting signed to sign-magnitude number

knezicm, 2024-11-12:
Realizovati kolo koje konvertuje označeni 8-bitni broj predstavljen u formi drugog komplementa u ekvivalentan 8-bitni označeni broj predstavljen u formatu znak sa magnitudom (*signed-magnitude*). Kod ovog formata, najviši bit je bit znaka, dok preostali biti predstavljaju apsolutnu vrijednost broja (npr. u ovom formatu broj "1111" umjesto -1 u formi drugog komplementa, ima vrijednost -7). Za realizaciju koristiti samo jedan 7-bitni komparator. Koristiti minimalan broj relacionih i aritmetičkih operatora u dizajnu. Entitet kola treba da ima sljedeći izgled:

```vhdl
entity signed_to_sign_magnitude_converter is
  port (
  SIGN_BIN_i : in  std_logic_vector (7 downto 0);
  SIGN_MAG_o : out std_logic_vector (7 downto 0)
);
end signed_to_sign_magnitude_converter;
```
gdje je `SIGN_BIN_i` ulazni broj, a `SIGN_MAG_o` izlazni konvertovani podatak.

Za realizovano kolo napisati odgovarajući samoprovjerljivi (*self-checking*) *testbench* neophodan za simulaciju kola. Naziv VHDL fajla za *testbench* treba da bude u formatu *nazivDizajna_tb.vhd*.

radislavkosijer, 2024-11-19:
Test prolazi za 255 vrijednosti, ali ne prolazi za ulaz -128, gdje očekuje izlaz 128, a kod mene se pojavljuje izlaz 0, ali u signed-magnitude reprezentaciji trebalo bi da dodam još jedan, deveti bit, da bih mogao predstaviti označeni broj 128, pa moja implementacija ima zapravo dvije 0, pozitivnu i negativnu, gdje mi je zapravo ta negativna označena nula ekvivalentna neoznačenom broju 128

knezicm, 2024-11-20:
Ideja je da se kolo realizuje na sljedeci nacin:

- bit znaka se prosto prenese sa ulaza ba izlaz
- ispita se status bita znaka, ako je 0 (broj je pozitivan) preostali nizi biti se prosto prenose na izlaz, a ako je 1 (broj je negativan), onda se primjenjuje konverzija negativnog u pozitivni broj metodom drugog komplementa, pri cemu se zadrzava bit znaka

Logika opisana iznad kao rezultat daje `10000000` na izlazu kada je ulazni podatak -128 (`10000000`). Iako ovo nije validan podatak, to je ocekivano, jer nije ni doveden validan ulazni podatak (minimalan broj u _signed-magnitude_ formatu je -127). Medjutim, na ovaj nacin dobijamo implementaciju koja se konzistentno moze prosirivati ukoliko je to potrebno (tj. nema nekih singulariteta u dizajnu).

radislavkosijer, 2024-11-20:
Izvinite, nisam vidio Vas komentar prije commita. 
To je bila takodje i moja prvobitna ideja, ali nisam pronasao nacin kako da iskoristim 7-bitni komparator, cija upotreba je naglasena u specifikaciji zadatka, pa mi je tu doslo do nejasnoce. 
U mom rijesenju, koje sam sad prilozio, ispravio sam uslov za -128 po uzoru na logiku koju ste mi predlozili, i u vhdl fajlu i u testbencu. Vidim da mi je sad test prosao, tako da, mislim da bi sada sve trebalo biti u redu.

Prilazem novi testbech_log
[testbench_log.txt](https://github.com/user-attachments/files/17833381/testbench_log.txt)

knezicm, 2024-11-23:
Razumijem da je konfuzija nastala zbog 7-bitnog komparatora koji je greskom dospio u tekst zadatka (vjerovatno kopiranjem dijela teksta iz nekog drugog zadatka). Popravicu tekst zadatka kako bi se ubuduce izbjegla ova situacija.

knezicm, 2025-01-29:
5 bodova

closed 2025-01-29 by knezicm