etf-unibl/pds-2024 #41: Create eight-bit multiplier

knezicm, 2024-11-12:
Realizovati 8-bitni množač neoznačenih brojeva `a` i `b`. Rezultat množenja se dobija na 16-bitnom izlazu `res`. Za realizaciju koristiti algoritam baziran na sabiranju parcijalnih proizvoda. Pri realizaciji koristiti pomjerače i sabirače. Pokušati optimizovati rješenje tako da se smanji broj sabirača potrebnih za realizaciju. Entitet kola treba da ima sljedeći izgled:

```vhdl
entity eight_bit_multiplier is
  port (
  A_i   : in  std_logic_vector(7 downto 0);
  B_i   : in  std_logic_vector(7 downto 0);
  RES_o : out std_logic_vector(15 downto 0)
);
end eight_bit_multiplier;
```
gdje su `A_i` i `B_i` ulazni 8-bitni brojevi koje množimo, a `RES_o` njihov proizvod.

Za realizovano kolo napisati odgovarajući samoprovjerljivi (*self-checking*) *testbench* neophodan za simulaciju kola. Naziv VHDL fajla za *testbench* treba da bude u formatu *nazivDizajna_tb.vhd*.

rankomirosav, 2024-12-20:
Log fajl fajl za ovaj issue ima u pull request koji se odnosi na ovaj zadatak

knezicm, 2025-01-29:
4 boda

closed 2025-01-29 by knezicm