etf-unibl/pds-2024 #42: Create Gray distance circuit

knezicm, 2024-11-12:
Realizovati kolo za određivanje udaljenosti *Gray* kodnih riječi. Ovu udaljenost definišemo kao broj tranzicija između kodnih riječi (npr. ako imamo 4-bitne *Gray* kodne riječi "0101" i "1111", udaljenost je 4, jer su potrebne četiri tranzicije - "0101"->"0100"->"1100"->"1101"->"1111"). Kolo realizovati u kompaktnoj formi bez korišćenja *brute-force* rješenja i sa mogućnošću proširenja na kodne riječi veće dužine. Entitet kola treba da ima sljedeći izgled:

```vhdl
entity gray_distance_unit is
  generic (
  g_N : natural := 4
);
  port (
  A_i        : in  std_logic_vector(g_N-1 downto 0);
  B_i        : in  std_logic_vector(g_N-1 downto 0);
  DISTANCE_o : out std_logic_vector(g_N-1 downto 0)
);
end gray_distance_unit;
```
gdje su `A_i` i `B_i` ulazni brojevi, a `DISTANCE_o` izračunata izlazna distanca. Generička konstanta `g_N` služi za definisanje dužine ulaznih podataka.

Za realizovano kolo napisati odgovarajući samoprovjerljivi (*self-checking*) *testbench* neophodan za simulaciju kola. Naziv VHDL fajla za *testbench* treba da bude u formatu *nazivDizajna_tb.vhd*.

jelena0000, 2024-11-19:
Potvrda o provjerama nakon pokretanja testova u lokalnom okruženju. 
[log.txt](https://github.com/user-attachments/files/17816383/log.txt)

knezicm, 2025-01-29:
5 bodova

closed 2025-01-29 by knezicm