etf-unibl/pds-2024 #43: Create Hamming distance circuit

knezicm, 2024-11-12:
Realizovati kolo koje određuje *Heming*ovu distancu između ulaznih 8-bitnih kodnih riječi `a` i `b`. Izračunata vrijednost se dobija na 4-bitnom izlazu `y`. Hemingova distanca predstavlja broj bitskih pozicija u kojima se dvije kodne riječi razlikuju (npr. kodne rijeći "00010011" i "10010010" imaju Hemingovu distancu dva, jer se razlikuju na poziciji 0 i 7). Kolo realizovati u kompaktnoj formi bez korišćenja *brute-force* rješenja i predvidjeti njegovu proširivost. Entitet kola treba da ima sljedeći izgled:

```vhdl
entity hamming_distance_unit is
  generic (
  g_WIDTH : natural := 8
);
  port (
  A_i : in  std_logic_vector(g_WIDTH-1 downto 0);
  B_i : in  std_logic_vector(g_WIDTH-1 downto 0);
  Y_o : out std_logic_vector(3 downto 0)
);
end hamming_distance_unit;
```
gdje su `A_i` i `B_i` ulazni brojevi, a `Y_o` izlazni podatak o distanci. Generička konstanta `g_WIDTH` određuje dužinu ulaznih brojeva.

Za realizovano kolo napisati odgovarajući samoprovjerljivi (*self-checking*) *testbench* neophodan za simulaciju kola. Naziv VHDL fajla za *testbench* treba da bude u formatu *nazivDizajna_tb.vhd*.

Neira7, 2024-11-17:
Potvrda o provjerama nakon pokretanja testova u lokalnom okruženju.
[log.txt](https://github.com/user-attachments/files/17791988/log.txt)

knezicm, 2025-01-29:
5 bodova

closed 2025-01-29 by knezicm