etf-unibl/pds-2024 #55: Create modified memory controller (second method)

knezicm, 2024-12-01:
Modifikovati kolo hipotetičkog kontrolera memorije, koje je opisano u video materijalu, tako da se omogući _back-to-back_ način rada, tj. da kontroler ne mora da se vraća u `idle` stanje nakon svake operacije, već da može da inicira novu operaciju odmah nakon završetka prethodne bez vraćanja u neaktivno stanje. U skladu s tim, potrebno je adaptirati mašinu sa konačnim brojem stanja i realizovati je opisanom metodologijom.

Entitet kola treba da ima sljedeći izgled:

```vhdl
entity mem_ctrl is
  port (
  clk_i   : in  std_logic;
  rst_i   : in  std_logic;
  mem_i   : in  std_logic;
  rw_i    : in  std_logic;
  burst_i : in  std_logic;
  oe_o    : out std_logic;
  we_o    : out std_logic;
  we_me_o : out std_logic
);
end mem_ctrl;
```

Za realizovano kolo napisati odgovarajući samoprovjerljivi _testbench_ neophodan za simulaciju kola.

Rješenje dokumentovati u skladu sa pravilima dokumentovanja VHDL opisa korišćenjem _Doxygen_ alata.

radislavkosijer, 2024-12-10:
U prilogu ostavljam ulazni csv fajl sa tabelom ocekivanih vrijednosti i izlazni csv fajl u kome su upisane i poredjene ocekivane i dobijene vrijednosti.
[mem_ctrl_input.csv](https://github.com/user-attachments/files/18082891/mem_ctrl_input.csv)
[mem_ctrl_output.csv](https://github.com/user-attachments/files/18082892/mem_ctrl_output.csv)

knezicm, 2025-01-29:
5 bodova

closed 2025-01-29 by knezicm