etf-unibl/pds-2024 #59: Create Manchester encoding circuit

knezicm, 2024-12-01:
Realizovati kolo koje na izlazu generiše sekvencu bita kodovanu Mačester kodom. Više o Mančester kodu možete pronaći na sljedećem linku: https://en.wikipedia.org/wiki/Manchester_code

Kod se može opisati dijagramom mašine stanja prikazanom na slici. Signal `d_i` predstavlja ulazne binarne podatke, dok signal `v_i` pokazuje da li je signal `d_i` validan (tj., da li postoje podaci koji se prenose). Jednostavnije rečeno, podaci u okviru signala `d_i` koduju se Mančester kod samo ako je signal `v_i` postavljen, inače treba da ostane na vrijednosti '0'.

[image]

Entitet kola treba da ima sljedeći izgled:

```vhdl
entity manchester_encoder is
  port (
  clk_i : in  std_logic;
  rst_i : in  std_logic;
  v_i   : in  std_logic;
  d_i   : in  std_logic;
  y_o   : out std_logic
);
end manchester_encoder;
```

Za realizovano kolo napisati odgovarajući samoprovjerljivi _testbench_ neophodan za simulaciju kola.

Rješenje dokumentovati u skladu sa pravilima dokumentovanja VHDL opisa korišćenjem _Doxygen_ alata.

Neira7, 2024-12-10:
Potvrda o provjerama nakon pokretanja testova u lokalnom okruženju. 
[log.txt](https://github.com/user-attachments/files/18076882/log.txt)

knezicm, 2025-01-29:
5 bodova

closed 2025-01-29 by knezicm