etf-unibl/pds-2024 #69: Create sequential divider using long-division algorithm

knezicm, 2024-12-19:
Realizovati sekvencijalno kolo za dijeljenje koje koristi tzv. _long-division_ algoritam (više informacija na linku: https://en.wikipedia.org/wiki/Long_division). U datom slučaju ulazni signali `a_i` i `b_i` predstavljaju dijeljenik i djelilac, respektivno, a izlazni signali su količnik `q_o` i ostatak `r_o`. Pretpostaviti da su svi signali dužine 8 bita i da predstavljaju neoznačene brojeve. Opisani algoritam realizovati RT metodologijom projektovanja.

Za slučaj dijeljenja nulom, rezultat kodovati kao `NaN` vrijednost koja odgovara vrijednosti 255 za količnik i ostatak.

Entitet kola treba da ima sljedeći izgled:

```vhdl
entity sequential_divider is
  port (
    clk_i   : in  std_logic;
    rst_i   : in  std_logic;
    start_i : in  std_logic;
    a_i     : in  std_logic_vector(7 downto 0);
    b_i     : in  std_logic_vector(7 downto 0);
    q_o     : out std_logic_vector(7 downto 0);
    r_o     : out std_logic_vector(7 downto 0);
    ready_o : out std_logic
);
end sequential_divider;
```

Rješenje dokumentovati u skladu sa pravilima dokumentovanja VHDL opisa korišćenjem _Doxygen_ alata.

U drugom dijelu zadatka, napraviti vremensku analizu projektovanog kola i specificirati vremenska ograničenja u vidu `.sdc` fajla i mapirati portove na stvarne pinove.

Neira7, 2025-01-23:
Izvjestaj o pregledanju zadatka. 

[Izvještaj.pdf](https://github.com/user-attachments/files/18520254/Izvjestaj.pdf)

knezicm, 2025-01-29:
5 bodova
Neira za review: 5 bodova

closed 2025-01-29 by knezicm