etf-unibl/pds-2024 #70: Create Fibonacci sequence generator

knezicm, 2024-12-19:
Realizovati sekvencijalno kolo za generisanje Fibonačijevih brojeva. Više o Fibonačijevoj funkciji, može se pronaći na linku: https://en.wikipedia.org/wiki/Fibonacci_number

Smatrati da neoznačeni 6-bitni ulazni broj `n_i` definiše dužinu Fibonačijevog niza. Opisani algoritam realizovati RT metodologijom projektovanja.

**Napomena:** Fibonačijev broj koji je 63. u nizu je 6557470319842.

Entitet kola treba da ima sljedeći izgled:

```vhdl
entity fibonacci is
  port (
      clk_i   : in  std_logic;
      rst_i   : in  std_logic;
      start_i : in  std_logic;
      n_i     : in  std_logic_vector(5 downto 0);
      r_o     : out std_logic_vector(42 downto 0);
      ready_o : out std_logic
);
end fibonacci;
```

Rješenje dokumentovati u skladu sa pravilima dokumentovanja VHDL opisa korišćenjem _Doxygen_ alata.

U drugom dijelu zadatka, napraviti vremensku analizu projektovanog kola i specificirati vremenska ograničenja u vidu `.sdc` fajla i mapirati portove na stvarne pinove.

Chicha13, 2025-01-24:
Izvjestaj:

[Komentar.txt](https://github.com/user-attachments/files/18541838/Komentar.txt)

Chicha13, 2025-01-24:
Testbench: 

[image]

Chicha13, 2025-01-24:
Vremenska analiza: 

-Unconstrained Paths
[image]

-Setup/Hold Time Slack
[image]

knezicm, 2025-01-31:
3 boda
Danilo za review: 5 bodova

closed 2025-01-31 by knezicm