etf-unibl/pds-2024 #71: Create leading zero counter circuit (FSMD approach)

knezicm, 2024-12-19:
Realizovati kolo za brojanje uzastopnih vodećih nula ulaznog 16-bitnog broja. Izvesti algoritam tako da se kolo može realizovati korišćenjem RT metodologije projektovanja.

Entitet kola treba da ima sljedeći izgled:

```vhdl
entity leading_zero_counter is
  port (
    clk_i   : in  std_logic;
    rst_i   : in  std_logic;
    start_i : in  std_logic;
    n_i     : in  std_logic_vector(15 downto 0);
    ready_o : out std_logic;
    zeros_o : out std_logic_vector(4 downto 0)
);
end leading_zero_counter;
```

Rješenje dokumentovati u skladu sa pravilima dokumentovanja VHDL opisa korišćenjem _Doxygen_ alata.

U drugom dijelu zadatka, napraviti vremensku analizu projektovanog kola i specificirati vremenska ograničenja u vidu `.sdc` fajla i mapirati portove na stvarne pinove.

rankomirosav, 2025-01-24:
U prilogu su dodani log falj koji potvrdjuje ispravnost dizajna kola za datim testbench-om, testbench za ovaj dizajn je dodan u okviru foldera u kom se nalaze sdc i dizaj kola fajlovi,  u prilogu se jos nalaze potrvrde da je vremenska analiza uredu i da kompajliranje prolazi.

[log_file.txt](https://github.com/user-attachments/files/18532791/log_file.txt)

[image]

[image]

knezicm, 2025-01-29:
4 boda
Ranko za review: 4 boda (zato sto nije uocio da coding standard nije ispostovan)