etf-unibl/pds-2024 #72: Create period counter

knezicm, 2024-12-19:
Projektovati kolo za mjerenje perioda ulaznog signala. Mjerenje se obavlja tako što se broje takt impulsi između dvije rastuće ivice na ulazu. Potrebno je obezbijediti mjerenje perioda u milisekundama. Izlazna vrijednost koja odgovara mjerenom periodu treba da bude 10-bitni neoznačeni broj. Kolo realizovati primjenom RT metodologije projektovanja.

Entitet kola treba da ima sljedeći izgled:

```vhdl
entity period_counter is
  port (
    clk_i    : in  std_logic;
    rst_i    : in  std_logic;
    signal_i : in  std_logic;
    period_o : out std_logic_vector(9 downto 0)
);
end period_counter;
```

Rješenje dokumentovati u skladu sa pravilima dokumentovanja VHDL opisa korišćenjem _Doxygen_ alata.

U drugom dijelu zadatka, napraviti vremensku analizu projektovanog kola i specificirati vremenska ograničenja u vidu `.sdc` fajla i mapirati portove na stvarne pinove.

jelena0000, 2025-01-24:
Izvještaj:

[Izvještaj.pdf](https://github.com/user-attachments/files/18540282/Izvjestaj.pdf)

knezicm, 2025-01-31:
3 boda
Jelena za review: 5 bodova

closed 2025-01-31 by knezicm